markruys/gw2pvo #2: Script needs adjusting - Changes to GoodWe website

Koploper, 2018-06-04:
Hi Mark,

I've been using your scripts for uploading my GoodWe PV generation data to PVOutput for the last few months. It worked conveniently. I especially found your tool to upload historical data very helpful.

A few days ago, due to the new privay legislation, Goodwe added a European server to deal with european data. It also seems that the URL to fetch the data does no longer work and that you need to log in before you can retrieve the data.

I have been looking for an alteranive and found one at https://github.com/zonoskar/Goodwe2PVoutput, which seems to work with the new server, but it lacks the tools to upload historical data. 

I f you have time, an update of gw2pvo would be greatly appriciated!

My best regards,

Koploper

markruys, 2018-06-04:
Ah, let me check this.

markruys, 2018-06-04:
Thanks for the link. I've update gw2pvo to use https (for both APIs), and added an option to set the region. In your case that would we `--gw-region EU`. Can you please test if it works for you too?

Koploper, 2018-06-04:
Hi Mark,

Works for me too.

Thanks!

Koploper

closed 2018-06-04 by markruys