markruys/gw2pvo #4: SEMS Portal

Sephyr0s, 2018-10-08:
Hi, so today my Goodwe user stopped working. I guess to force us to the new SEMS Portal. I guess unless you're going to modify it to work with SEMS it's going to make this script obsolete.

markruys, 2018-10-08:
Most likely we need PR https://github.com/markruys/gw2pvo/pull/3. I'll try to merge it soon now.

markruys, 2018-10-08:
I've release [1.2.0](https://github.com/markruys/gw2pvo/releases/tag/1.2.0), which should address the issue. Please let us now how it works for you.

peternijssen, 2018-10-09:
For me it kept working on 1.1.0 (for whatever reason). However, I just updated to 1.2.1 and everything seems to work fine as far as I can see right now. Just waiting for the sun to shine ;)

Thanks Mark for your work on this tool!

Sephyr0s, 2018-10-10:
Cheers, installed it and it works like a charm. :)

closed 2018-10-15 by markruys