markruys/gw2pvo #8: All imported logs with --date are 1 day ahead

robinbohnen, 2018-12-14:
When I import with this command: gw2pvo --gw-station-id <station> --gw-account <account> --gw-password <pw> --pvo-system-id <id> --pvo-api-key <key> --log debug --date 2018-12-13 --darksky-api-key <key>

It will import the data of 2018-12-12 (at GoodWe) to 2018-12-13 (at Pvoutput).
How can I fix this?

eNeRGy164, 2018-12-15:
I think this is because the timezone code has been changed/removed in the latest commit?

markruys, 2018-12-17:
Might be very well the case. The original code gave errors (cf issue #7), so we have to look into this again.

vijghph, 2019-01-19:
I confirm the issue: the retroactive loading of data is setup day "one+1". To be fixed.

vijghph, 2019-01-20:
I noticed the JSON response for GetPowerStationPowerAndIncomeByDay() contains a sequence of data for day-1 then the day itself. So the fix maybe actually quite simple. I'll propose it.

Koploper, 2019-02-11:
I confirm the issue also. Can the fix be added to the code?

vijghph, 2019-03-24:
Any hope of getting a build with the proposed fix?

daemenseth, 2019-03-25:
I can confirm the fix is working. I hope there is fast a build with the fixes

lbbrhzn, 2019-04-05:
the fix from vijghph is working for me as well.

lbbrhzn, 2019-04-05:
Unfortunately, this fix is does seem to work for the current day.

The issue appear to be that GetPowerStationPowerAndIncomeByDay does not return the running total for the current day. 

However,  GetPowerStationPacByDayForApp appears to have this info in the 'today_power' field.  Note that it always contains the value for the *current* day, not for the requested day!!

commit d894b02d (2020-06-28): Fix invocation with --date (fixes #8, #11, #14, #25, #41)
closed 2020-06-28 by markruys