markruys/gw2pvo #9: Only one inverter data is counted for GW_STATION_ID

vijghph, 2019-01-13:
My GoodWe plant is composed of *two* inverters but the tool only upload to pvoutput the data of one of them. It expect it to iterate over all inverters of the plant and sum the figures up (as is doing https://github.com/buttonfreak/goodwe-api)

For the rest, the tool is very sophisticated and working gracefully. Moreover the existence of a docker package let it install it ike a charm.

eNeRGy164, 2019-01-13:
As a note, when changing to the new API, I saw it would be easy to iterate through the list of invertors. Also removing the required parameter for the application. So it shouldn't be hard to add this to the script.
At that moment I didn't want to change that part of the script as it was not related to supporting the new API.

vijghph, 2019-01-19:
I just proposed a file changed that I have properly tested.

Can you validate it to get it approved.
I hope it would eventually go also to the Docker image.

vijghph, 2019-03-24:
Any hope to get a build with the proposed fix?

vijghph, 2019-06-17:
Was it included in today build? If not, what should I do to get it released?

wiljums, 2019-11-06:
From which inverter is the data uploaded currently? And is it possible to get the data of the other inverter? I have currently only offline readings, which could come from my old (offline) inverter. The data of the new one isn't uploaded..

markruys, 2020-06-29:
I close this ticket as it's a duplicate of #19. In #19 I've added a reference to a feature branch which I'll merge if someone with multiple inverters can test the change.

markruys, 2020-06-30:
I was under the impression that the request for support for multiple strings and multiple inverters were basically the same. Which is not, so I reopen this ticket, as this one is about multiple inverters. 

I've created a [feature branch](https://github.com/markruys/gw2pvo/tree/feature/multiple_inverters) which should do the trick. I have only a single inverter, so can someone test this change?

bramhut, 2020-07-02:
I would be able to test it, however I'm not sure how to install the feature branch. Can you give me some hints maybe?
I'm still on version 1.2.1 and running it in docker on Windows 10.

markruys, 2020-07-05:
But I've merged this feature into trunk and released it as 1.3.4. You can try it like:

```
sudo pip3 install gw2pvo==1.3.4
```

closed 2020-07-05 by markruys

bramhut, 2020-07-07:
Okay thanks, it works good now!