markruys/gw2pvo #14: When using the option --date "on the current day", total production is incorrect

vijghph, 2019-03-24:
When using the --day flag to load the historical data for "today", the total production in kph is incorrect because it is actually the one returned for "yesterday" by the GoodWe API.

In such case, I would recommend instead to just sum up incrementally the production data (because the daily total is computed asynchronously in batch mode by GoodWe server), instead of trying to normalise it based on GoodWe calculations.

Steve-Wax, 2019-07-13:
I noticed the same problem and found that is was linked to the correction factor that is being calculated in the gw_api.py, getDayReadings function.  If the correction factor worked properly it might make the total production more accurate, but I haven't looked into that.  I found that if I ignore the correction factor it is good enough for me.  So what I changed this line in the gw_api.py, getDayReadings function:

`increase = pgrid_w * sample['minutes'] * factor`

to this:

`increase = pgrid_w * sample['minutes'] / 1000 / 60`

This change worked for me and seems pretty accurate for total production.

Steve-Wax, 2019-07-13:
When looking into issue #23 I noticed that the return from /api/PowerStationMonitor/GetPowerStationPowerAndIncomeByDay is two days of total power value.  It includes the date passed in the --date parameter and the previous day:

> 'data': [{'i': 1.1573, 'd': '07/11/2019', 'p': 16.3}, {'i': 1.6614, 'd': '07/12/2019', 'p': 23.4}]

The script assigns the power for the day using this code:

`eday_kwh = data[0]['p']`

I seems to me that the prior day's power is used rather than the current days value.  Assuming the GoodWe return data is consistent the code needs to change to:

`eday_kwh = data[1]['p']`

Steve-Wax, 2019-07-13:
I've done some further investigation and found some additional changes that need to be made to correct this problem.

1. As mentioned above the previous day's power is being used.  Correction as per above.

2. The units are not correct when determining the correction factor.  The eday_from_power calculation shoud be: `eday_from_power += sample['pac'] * sample['minutes'] / 60 / 1000` to keep this value in kWh as per the return of the daily power.

3. The units are not correct when calculating the eday_kWh increase.  The increase calculation should change to: `increase = pgrid_w * sample['minutes'] * factor /60 /1000`.

There maybe a better way to do this, but this is what makes sense to me.  I've tested it out on one day of my data and works, although I haven't implemented it in my production script yet.  So with all the changes this section of code would look like this:

`       
        eday_kwh = data[1]['p']

        payload = {
            'id' : self.system_id,
            'date' : date_s
        }
        data = self.call("PowerStationMonitor/GetPowerStationPacByDayForApp", payload)
        if 'pacs' not in data:
            logging.warning(date_s + " - Received bad data " + str(data))
            return result

        minutes = 0
        eday_from_power = 0
        for sample in data['pacs']:
            parsed_date = datetime.strptime(sample['date'], "%m/%d/%Y %H:%M:%S")
            next_minutes = parsed_date.hour * 60 + parsed_date.minute
            sample['minutes'] = next_minutes - minutes
            minutes = next_minutes
            eday_from_power += sample['pac'] * sample['minutes'] / 60 / 1000
        factor = eday_kwh / eday_from_power if eday_from_power > 0 else 1

        eday_kwh = 0
        for sample in data['pacs']:
            date += timedelta(minutes=sample['minutes'])
            pgrid_w = sample['pac']
            increase = pgrid_w * sample['minutes'] * factor
            if increase > 0:
                eday_kwh += increase
                result['entries'].append({
                    'dt' : date,
                    'pgrid_w': pgrid_w,
                    'eday_kwh': round(eday_kwh, 3))}`

Steve-Wax, 2019-07-14:
I realized that the first correction above is not a particularly robust solution.  Instead of `eday_kwh = data[1]['p']` it would be better to look for the correct day in the data.  So instead I think it should be:

```
convertedDate = date_s[5:7] + "/" + date_s[-2:] + "/" + date_s[:4]
eday_kwh = 0
for day in data:
    if day['d'] == convertedDate:
        eday_kwh = day['p']
```
This way in case there days are out of order or they decide to pass more days we can get the correct power.

Steve-Wax, 2019-07-27:
I believe I have not addressed all the issues with using the --day option.  Note that you need to wait for awhile after a day is over for this option to work.  I'm using it at 4:30 am the next day and have good results.  I will also add a --day yesterday option to make this easier to use on a daily script.  I'm running a crontab daily to get the prior days readings to pvoutput.  I also found that the power curve was starting at the correct time of day and that was fixed in the pull request that is referenced above.

markruys, 2020-06-28:
Should be fixed in release 1.3.2.

closed 2020-06-28 by markruys