markruys/gw2pvo #16: the first value for the day is too high

daemenseth, 2019-03-26:
I using this project since yesterday but todays first import is too high
it is 1800

In the logging I see also the v1 is 1800
What can cause this issue.
I have implemted the fixes from @vijghph 
https://pvoutput.org/intraday.jsp?id=74365&sid=65959&dt=20190326

X-gen76, 2019-04-25:
I got the same problem since 7 days ago.

Ralf7, 2019-04-27:
Since April 19 I've got this issue too. See https://pvoutput.org/intraday.jsp?id=62026&sid=55213
I think something has changed on the GoodWe server, because I haven't changed anything. 
I hope somebody has got a solution for this, as my solar array is facing west, I can see the real values only from 1PM on.

walterwhite81, 2019-04-30:
Same here, starts with 4.550kWh every day.

vkevin85, 2019-05-01:
Same problem here, however de starting value of the day is different from day to day. Already applied the fix from [vijghph](https://github.com/markruys/gw2pvo/pull/11) but it doesn't seem to make any difference. Hopefuly someone can help to fix this issue.

tclaer, 2019-05-06:
same issue....

X-gen76, 2019-05-10:
For the people how wants/needs an alternative:

https://github.com/zonoskar/Goodwe2PVoutput/

Simanias, 2019-05-24:
To bad I'm having the same issue.
Besides this isseu, the script works great.

JaapBraam, 2019-06-07:
Today this happened to me too... Yesterday I added the --skip-offline option. Maybe related?

JaapBraam, 2019-06-09:
I investigated: 
I adapted the code to not use an average for the generated power, and now production starts at 0 kwh the next day.... I'll make a pull request for this...

vkevin85, 2019-06-12:
Pulled the changes you made Jaap and it seems to work fine, thanks for your effort.

X-Wizard, 2019-06-12:
Hi Jaap, I modified the script and it works, thanks!

2 Question,
- How to deal with two MPPT, is there a possibility to sum both MPPT?
- How to use the temperature value from the inverter?

markruys, 2019-06-13:
Looks like people are happy about  #PR18, so I accepted it. Thank Jaap for you effort!

walterwhite81, 2019-06-14:
This fixed it, thanks @JaapBraam !

Ralf7, 2019-06-16:
How can I apply this fix? 
I’ve tried to reinstall the package, but I received notifications of duplicate files. Then reinstalled the package with —upgrade at the end. That resulted in some errors. 
Can anybody describe in easy-to-follow steps what to do?

eNeRGy164, 2019-06-16:
As it's not a new release yet, you would need to build the package from source code yourself

Ralf7, 2019-06-17:
Ok that’s a pity, in that case I will wait until a new version comes out.

markruys, 2019-06-17:
I'll build a new release this evening (if don't forget it ;)

markruys, 2019-06-17:
Build new release: https://github.com/markruys/gw2pvo/releases/tag/1.2.2

Ralf7, 2019-06-22:
I installed the new release 3 days ago and it’s working like a charm! Thanks for the effort @JaapBraam and @markruys

closed 2019-06-23 by markruys