markruys/gw2pvo #17: inverter temperature not used

lbbrhzn, 2019-04-05:
gw2pvo version 1.2.1 uses the darksky local temperature info to populate the temperature that is uploaded to pvoutput. That works fine, but I'd rather see the inverter temperature instead.

For the live status updates this can be retrieved from the GetMonitorDetailByPowerstationId query result: There is a field called 'tempperature' (Note the spelling mistake!)

Don't know how to obtain all temperatures for a particular day though. The sems web ui can export this data through a report.

ozblogger, 2020-01-30:
Thanks for that tip. I used it to monitor the temperature of the inverter
Replace Line 42 in gw_api.py  with the following, and it will do the job:

`            'longitude' : data['info'].get('longitude'),
            'temperature' : inverterData['tempperature'] `

markruys, 2020-06-29:
IMHO we should use the environmental temperature in PVOutput, not the (often much much higher) internal inverter temperature. But of course feel free to fork & hack this repo as you wish.

closed 2020-06-29 by markruys