markruys/gw2pvo #19: Inverter with dual string

X-Wizard, 2019-06-13:
Gents,

I have an inverter with a duel string vpv1 and vpv2 (voltage), is there a possibility to sum both strings?

X-Wizard, 2019-06-20:
Would be nice if we can sum both or more strings or has someone else a solution to tackle this?

markruys, 2020-06-29:
I've created a [feature branch](https://github.com/markruys/gw2pvo/tree/feature/multiple_inverters) which should do the trick. I have only a single inverter, so can someone test this change?

X-Wizard, 2020-06-30:
Hi Markruys,

I changed both files as requested, the output is 1 PV string shown on pvoutput.org.

markruys, 2020-06-30:
Strange, can you show me the output of your `gw2pvo` invocation with the `--log debug` added?

X-Wizard, 2020-06-30:
Hi Markruys,

Please find included in this email the requested data,

-- 

Jun 30 10:28:13 vps gw2pvo[18459]: DEBUG    Starting new HTTPS connection (1): semsportal.com:443
Jun 30 10:28:13 vps gw2pvo[18459]: DEBUG    https://semsportal.com:443 "POST /api/v2/PowerStation/GetMonitorDetailByPowerstationId HTTP/1.1" 200 300
Jun 30 10:28:13 vps gw2pvo[18459]: DEBUG    {'hasError': False, 'code': 100001, 'msg': 'No access, please login.', 'data': None, 'components': {'para': None, 'langVer': 73, 'timeSpan': 0, 'api': 'http://semsportal.com:82/api/v2/PowerStation/GetMonitorDetailByPowerstationId', 'msgSocketAdr': ''}}
Jun 30 10:28:13 vps gw2pvo[18459]: DEBUG    Starting new HTTPS connection (1): semsportal.com:443
Jun 30 10:28:13 vps gw2pvo[18459]: DEBUG    https://semsportal.com:443 "POST /api/v2/Common/CrossLogin HTTP/1.1" 200 527
Jun 30 10:28:14 vps gw2pvo[18459]: DEBUG    Starting new HTTPS connection (1): eu.semsportal.com:443
Jun 30 10:28:15 vps gw2pvo[18459]: DEBUG    https://eu.semsportal.com:443 "POST /api/v2/PowerStation/GetMonitorDetailByPowerstationId HTTP/1.1" 200 25114
Jun 30 10:28:15 vps gw2pvo[18459]: DEBUG    {'language': 'en', 'function': ['ADD', 'VIEW', 'EDIT', 'DELETE', 'INVERTER_A', 'INVERTER_E', 'INVERTER_D'], 'hasError': False, 'msg': 'success', 'code': '0', 'data': {'info': {'powerstation_id': '6bc75339-f48c-440f-a096-a1dc6bf74574', 'time': '06/30/2020 10:28:15', 'date_format': 'MM.dd.yyyy', 'date_format_ym': 'MM.yyyy', 'stationname': 'JP Energy House', 'address': 'A, Nederland ', 'owner_name': None, 'owner_phone': None, 'owner_email': '', 'battery_capacity': 0.0, 'turnon_time': '05/16/2019 00:00:00', 'create_time': '05/16/2019 01:12:55', 'capacity': 3.0, 'longitude':, 'latitude':, 'powerstation_type': 'residential', 'status': 1, 'is_stored': False, 'is_powerflow': False, 'charts_type': 4, 'has_pv': True, 'has_statistics_charts': False, 'only_bps': False, 'only_bpu': False, 'time_span': -2.0, 'pr_value': ''}, 'kpi': {'pac': 238.0, 'power': 0.3, 'total_power': 4081.4, 'day_income': 0.07, 'total_income': 914.23, 'yield_rate': 0.224, 'currency': 'EUR'}, 'images': ['https://oversea-gops.oss-cn-hongkong.aliyuncs.com/powerstation_img/1c5b6e89-c109-41b6-a5f0.jpeg', 'https://oversea-gops.oss-cn-hongkong.aliyuncs.com/powerstation_img/a8b8bf6d-0a31-4143-9a66.jpeg'], 'weather': {'HeWeather6': [{'daily_forecast': [{'cond_code_d': '300', 'cond_code_n': '300', 'cond_txt_d': 'Shower Rain', 'cond_txt_n': 'Shower Rain', 'date': '2020-06-30', 'time': '2020-06-30 00:00:00', 'hum': '90', 'pcpn': '4.8', 'pop': '67', 'pres': '1007', 'tmp_max': '19', 'tmp_min': '16', 'uv_index': '7', 'vis': '24', 'wind_deg': '245', 'wind_dir': 'SW', 'wind_sc': '3-4', 'wind_spd': '24'}, {'cond_code_d': '302', 'cond_code_n': '101', 'cond_txt_d': 'Thundershower', 'cond_txt_n': 'Cloudy', 'date': '2020-07-01', 'time': '2020-07-01 00:00:00', 'hum': '80', 'pcpn': '4.8', 'pop': '60', 'pres': '1006', 'tmp_max': '21', 'tmp_min': '15', 'uv_index': '4', 'vis': '24', 'wind_deg': '235', 'wind_dir': 'SW', 'wind_sc': '3-4', 'wind_spd': '16'}, {'cond_code_d': '300', 'cond_code_n': '100', 'cond_txt_d': 'Shower Rain', 'cond_txt_n': 'Clear', 'date': '2020-07-02', 'time': '2020-07-02 00:00:00', 'hum': '80', 'pcpn': '4.8', 'pop': '67', 'pres': '1012', 'tmp_max': '19', 'tmp_min': '12', 'uv_index': '6', 'vis': '24', 'wind_deg': '315', 'wind_dir': 'NW', 'wind_sc': '1-2', 'wind_spd': '10'}, {'cond_code_d': '101', 'cond_code_n': '101', 'cond_txt_d': 'Cloudy', 'cond_txt_n': 'Cloudy', 'date': '2020-07-03', 'time': '2020-07-03 00:00:00', 'hum': '77', 'pcpn': '0.0', 'pop': '25', 'pres': '1017', 'tmp_max': '20', 'tmp_min': '15', 'uv_index': '6', 'vis': '24', 'wind_deg': '229', 'wind_dir': 'SW', 'wind_sc': '3-4', 'wind_spd': '20'}, {'cond_code_d': '300', 'cond_code_n': '300', 'cond_txt_d': 'Shower Rain', 'cond_txt_n': 'Shower Rain', 'date': '2020-07-04', 'time': '2020-07-04 00:00:00', 'hum': '81', 'pcpn': '2.0', 'pop': '60', 'pres': '1015', 'tmp_max': '20', 'tmp_min': '13', 'uv_index': '2', 'vis': '19', 'wind_deg': '212', 'wind_dir': 'SW', 'wind_sc': '4-5', 'wind_spd': '26'}, {'cond_code_d': '300', 'cond_code_n': '103', 'cond_txt_d': 'Shower Rain', 'cond_txt_n': 'Partly Cloudy', 'date': '2020-07-05', 'time': '2020-07-05 00:00:00', 'hum': '78', 'pcpn': '0.6', 'pop': '55', 'pres': '1013', 'tmp_max': '17', 'tmp_min': '12', 'uv_index': '4', 'vis': '24', 'wind_deg': '257', 'wind_dir': 'SW', 'wind_sc': '4-5', 'wind_spd': '25'}, {'cond_code_d': '999', 'cond_code_n': '999', 'cond_txt_d': '-', 'cond_txt_n': '-', 'date': '2020-07-06', 'time': '2020-07-06 00:00:00', 'hum': '-', 'pcpn': '-', 'pop': '-', 'pres': '-', 'tmp_max': '-', 'tmp_min': '-', 'uv_index': '-', 'vis': '-', 'wind_deg': '-', 'wind_dir': '--', 'wind_sc': '--', 'wind_spd': '-'}], 'basic': {'cid': 'NL2759879', 'location': 'Almere-Stad', 'cnty': 'Netherlands', 'lat': '52.36700058', 'lon': '5.21799994', 'tz': '+1.00'}, 'update': {'loc': '2020-06-29 22:49:00', 'utc': '2020-06-29 21:49:00'}, 'status': 'ok'}]}, 'inverter': [{'sn': '93000', 'dict': {'left': [{'key': 'Model', 'value': 'GW3000D-NS', 'unit': '', 'isFaultMsg': 0, 'faultMsgCode': 0}, {'key': 'S/N', 'value': '93000', 'unit': '', 'isFaultMsg': 0, 'faultMsgCode': 0}, {'key': 'Checkcode', 'value': '024982', 'unit': '', 'isFaultMsg': 0, 'faultMsgCode': 0}, {'key': 'Capaciteit', 'value': '3', 'unit': 'kW', 'isFaultMsg': 0, 'faultMsgCode': 0}, {'key': 'Connected', 'value': '05.15.2019 20:59:09', 'unit': '', 'isFaultMsg': 0, 'faultMsgCode': 0}, {'key' [... cut ...]
Jun 30 10:28:15 vps gw2pvo[18459]: DEBUG    https://api.darksky.net:443 "GET /forecast/00055055a01b1a5ccfe08565cd317f3b/52.374063,5.241989?units=si&exclude=minutely,hourly,daily,alerts,flags HTTP/1.1" 200 435
Jun 30 10:28:15 vps gw2pvo[18459]: DEBUG    {'d': '20200630', 't': '10:28', 'v1': 300, 'v2': 238, 'v5': 17.3, 'v6': 227.1}
Jun 30 10:28:15 vps gw2pvo[18459]: DEBUG    Starting new HTTPS connection (1): pvoutput.org:443
Jun 30 10:28:16 vps gw2pvo[18459]: DEBUG    https://pvoutput.org:443 "POST /service/r2/addstatus.jsp HTTP/1.1" 200 None

markruys, 2020-06-30:
Now I understand we're talking about two different requests:
* support for multiple inverters
* support for multiple strings

So let's keep this issue to sum voltages for multiple strings. That's an easy change, I'll implement it.

X-Wizard, 2020-06-30:
Correct, my request was support for multiple string.

X-Wizard, 2020-06-30:
The modification for the sum (voltage) of both strings works perfectly!

Thanks!

closed 2020-06-30 by X-Wizard