markruys/gw2pvo #20: Energy does not reset each day

JWDebler, 2019-06-28:
Yes, I am running the Bugfix Release 1.2.2.

There appears to be some power being reported, which doesn't reset the energy to 0 and I start the next day with yesterday's result (6.50 kWh was the output of the previous day).
Maybe we can get a flag to hard reset to 0 at midnight? 
[image]

kingfisher02, 2019-07-22:
I had the same issue, but it got fixed in version 1.2.2.

JWDebler, 2019-08-23:
sometimes it works, sometimes it doesn't

eNeRGy164, 2019-08-27:
As far as I remember, running total is calculated by PVOutput.

Do you use the `--skip-offline` switch? The GoodWe API keeps reporting the last measurements, even when the inverter is offline.

TheHawkAU, 2019-08-27:
I've noticed this behaviour too, although in my case it's half of yesterdays total.  I am assuming it's the average of the last reading from yesterday + 0 from the first reading of the day.

Which means my graph looks like this with a high total until the new total for today for the day exceeds that number.

[image]

Do you have to provide that running total as part of the upload or will PVOutput calculate it for you?

TheHawkAU, 2019-08-27:
I hadn't before, but I have now, just rebooted to ensure everything is applied and we will see how things go over the next couple of days.  For the moment it appears to have stopped logging.

TheHawkAU, 2019-08-29:
**Update:**

**26/08** - the last reading of the day was repeated until midnight, then the following day started at half the previous days energy total.  Once this was exceeded things continued normally.

**27/08** - Again started with hald the previous days energy total.  That evening I added in the --skip-offline command and rebooted the machine and it stopped showing records into the evening once the inverter went offline.

**28/08** - Looks like success!  The first energy reading for the day is 0.1 and it goes up over the course of the day again stops reporting once the inverter goes offline in the evening.

**29/08 (Today)** - Back to reporting half of the previous days reading at the energy total).
https://pvoutput.org/intraday.jsp?id=47592&sid=61960&dt=20190829

So the switch stopped all the late-night entries but doesn't appear to have resolved the underlying energy issue (was it just the reboot that did that?).  Will see again tomorrow.

I wonder if this is still something to do with averaging between the last entry and the first for the day?  Maybe with skipping offline there isn't then a reset to 0 at midight?  Will have to look into the code a little more when I have some spare time.

markruys, 2020-06-29:
I've added `--skip-offline` to the Systemd service example in the README.md.

closed 2020-06-29 by markruys