markruys/gw2pvo #21: change to API URL?

JWDebler, 2019-07-08:
I have been getting this error since earlier today:

`Jul 08 18:37:20 odroid gw2pvo[7181]: WARNING  HTTPSConnectionPool(host='globalapi.sems.com.cn', port=443): Max retries exceeded with url: /api/v1/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xb5c95a10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',))
Jul 08 18:37:47 odroid gw2pvo[7181]: ERROR    Failed to call GoodWe API
Jul 08 18:37:47 odroid gw2pvo[7181]: ERROR    'inverter'`

It looks like `globalapi.sems.com.cn` is not reachable anymore. Has goodwe maybe changed or removed that URL? I can see all my stats just fine on semsportal.

walterwhite81, 2019-07-08:
Same-ish- here: Jul  8 13:29:04 host gw2pvo[12799]: WARNING  HTTPSConnectionPool(host='globalapi.sems.com.cn', port=443): Max retries exceeded with url: /api/v1/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f1fe99c58d0>: Failed to establish a new connection: [Errno -2] Name or service not known'))

RubenR, 2019-07-08:
Same here:
INFO     Starting new HTTPS connection (1): globalapi.sems.com.cn
WARNING  ('Connection aborted.', gaierror(-2, 'Name or service not known'))

Uma1r, 2019-07-08:
Yeah, DNS is not resolving.

JaapBraam, 2019-07-08:
New server name is https://euapi.sems.com.cn/api/

walterwhite81, 2019-07-08:
verified, globalapi.sems.com.cn working fine again.

euapi doesn't resolve for me.

closed 2019-07-09 by JWDebler