markruys/gw2pvo #23: "Basic SEMS portal login failed" DSM update?

bldewit, 2019-07-12:
Hi,

After using this script for several months successfully, since yesterday I get this login error. Only thing I can think of tha5 has changed is a DSM update on my Synology NAS.

In Debug mode I didn't find a clue:

> 
ash-4.3# /bin/php goodwe2pvo.php

<html>

<header>

</header>

<body>

<pre>

array(6) {

  ["hasError"]=>

  bool(false)

  ["code"]=>

  int(0)

  ["msg"]=>

  string(7) "Success"

  ["data"]=>

  array(6) {

    ["uid"]=>

    string(36) "ac826737-1ee6-45b5-b4d7-ef5e428700d2"

    ["timestamp"]=>

    float(1562924506410)

    ["token"]=>

    string(32) "84ac808a7306734bf6610854d25a9910"

    ["client"]=>

    string(3) "ios"

    ["version"]=>

    string(6) "v2.1.0"

    ["language"]=>

    string(2) "en"

  }

  ["components"]=>

  array(4) {

    ["para"]=>

    NULL

    ["langVer"]=>

    int(25)

    ["timeSpan"]=>

    int(0)

    ["api"]=>

    string(45) "http://eu.semsportal.com:82/api/Auth/GetToken"

  }

  ["api"]=>

  string(30) "https://eu.semsportal.com/api/"

}

<br/>Basic SEMS portal login failed. Enable debug mode 

> 

Any suggestions?

Steve-Wax, 2019-07-13:
Mine was failing also, but I don't think it has anything to do with DSM.  When reviewing the debug output of the gw2pvo call I noticed that the calls were successful, but the script was acting like they failed.  After looking around a little I noticed that GoodWe had changed the response message from "success" to "Success".

> DEBUG    {'data': [{'i': 1.1573, 'd': '07/11/2019', 'p': 16.3}, {'i': 1.6614, 'd': '07/12/2019', 'p': 23.4}], 'hasError': False, 'components': {'timeSpan': 0, 'langVer': 25, 'para': None, 'api': 'http://au.semsportal.com:82/api/PowerStationMonitor/GetPowerStationPowerAndIncomeByDay'}, 'msg': 'Success', 'code': 0}

In the gw_api.py file, send function, I changed the check for success to use the python lower() function and after that it worked.
```

> if data['msg'].lower() == 'success' and data['data'] is not None:
                    return data['data']
```

vijghph, 2019-07-13:
Indeed: that was the issue!! Your trick (.lower()) works well. Thanks

TheHawkAU, 2019-08-07:
I was running this in Centos and hit an error at about the same time and saw this today, turns out it fixed my issue too.  Thanks.

groentje, 2019-08-08:
@Steve-Wax thank you!
in the file:  /usr/local/lib/python3.7/dist-packages/gw2pvo/gw_api.py
edit line 128 : ` if data['msg'] == 'success' and data['data'] is not None:`
to:  `if data['msg'].lower() == 'success' and data['data'] is not None:`

markruys, 2020-06-28:
I've switch to checking data['code'] instead, which should work more reliable (4b97b7a).

closed 2020-06-28 by markruys

markruys, 2020-06-28:
I've just released it as a new package:
```
pip3 install https://github.com/markruys/gw2pvo/releases/download/1.3.1/gw2pvo-1.3.1-py3-none-any.whl
```