markruys/gw2pvo #28: Uploading history error

groentje, 2019-08-05:
Hi Mark,

When i try to upload history data i get an error:
ERROR    Failed to call GoodWe API
WARNING  2019-08-05 - Received bad data {}
Command i use is:
gw2pvo --gw-station-id ** --gw-account ** --gw-password ** --pvo-system-id ** --pvo-api-key ** --date 2019-08-05
This command did work for the last time on 2019-07-02

I also have a service running and this service is running without any problems. Same command but without -- date

OS: Lubuntu
Python 3.7.3

Thanks!

Steve-Wax, 2019-08-07:
You may be have the same problem as described in Issue #23.

groentje, 2019-08-08:
@Steve-Wax thank you!
in the file:  /usr/local/lib/python3.7/dist-packages/gw2pvo/gw_api.py
edit line 128 : ` if data['msg'] == 'success' and data['data'] is not None:`
to:  `if data['msg'].lower() == 'success' and data['data'] is not None:`

closed 2019-08-08 by groentje