markruys/gw2pvo #34: msg response field is inconsistent on API

agvaldezc, 2019-10-06:
https://github.com/markruys/gw2pvo/blob/878d3289444f05b5281d37bca4013eec0722fc3a/gw2pvo/gw_api.py#L129

The API response on has different values on `msg` field sometimes have the `success` and other have `Success` so the lower and upper case cuases it to crash.

This is the solution I have on my local environment:
```python
                msg = data['msg']

                if msg.lower() == 'success' and data['data'] is not None:
                    return data['data']
```

Steve-Wax, 2019-10-06:
Same as Issue #23.

markruys, 2020-06-28:
I've switch to checking data['code'] instead, which should work more reliable (4b97b7a9e7acad0030b01e555603cc5440ab8be6).

closed 2020-06-28 by markruys

markruys, 2020-06-28:
I've just released it as a new package:
```
pip3 install https://github.com/markruys/gw2pvo/releases/download/1.3.1/gw2pvo-1.3.1-py3-none-any.whl
```