markruys/gw2pvo #36: Max retires exceeded.

Nevicare, 2020-01-30:
After installing gw2pvo I get the error:

HTTPSConnectionPool(host='globalapi.sems.com.cn', port=443): Max retries exceeded with url: /api/v1/PowerStation/GetMonitorDetailByPowerstationId (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f7a697ab978>: Failed to establish a new connection: [Errno -3] Try again',)

When I connect to globalapi.sems.com.cn I see the available API commands.
The command PowerStation/GetMonitorDetailByPowerstationId is not shown.

(I already changed the gw_api.py with data['msg'].lower() == 'success'.)

Is this a known issue?

franksnl, 2020-06-09:
I had this issue when using Docker on CentOS 8 with FirewallD.
Fixed by applying the following firewall rules:

firewall-cmd --permanent --zone=public --add-masquerade 
​firewall-cmd --permanent --zone=trusted --change-interface=docker0
firewall-cmd --permanent --zone=trusted --add-port=4243/tcp
firewall-cmd --reload

markruys, 2020-06-28:
Also you might want to try this update:
```
sudo pip3 install https://github.com/markruys/gw2pvo/releases/download/1.3.2/gw2pvo-1.3.2-py3-none-any.whl
```

markruys, 2020-06-30:
Please let me know if the latest release still has this issue. I expect it to be solved, so I close this issue for now.

closed 2020-06-30 by markruys