markruys/gw2pvo #37: ImportError: cannot import name 'Astral' from 'astral'

pumpkins001, 2020-02-23:
Hi, just tried to get this program running and I get the following error. Note all sensitive info has been replaced. Any assistance would be appreciated. I have Astral installed (v2.1) Python version:

Python 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)] on win32

Usage and error:

C:\>gw2pvo --gw-station-id xxxxxxxxxxxxx --gw-account xxxxxxxxxx--gw-password xxxxxxxxxxx--pvo-system-id xxxxxx --pvo-api-key xxxxxxxxxxxxxx --log debug
Traceback (most recent call last):
  File "C:\Users\yyyyyyyyyy\AppData\Local\Programs\Python\Python38-32\Scripts\gw2pvo-script.py", line 11, in <module>
    load_entry_point('gw2pvo==1.2.1', 'console_scripts', 'gw2pvo')()
  File "c:\users\yyyyyyyyyy\appdata\local\programs\python\python38-32\lib\site-packages\pkg_resources\__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\users\yyyyyyyyyy\appdata\local\programs\python\python38-32\lib\site-packages\pkg_resources\__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "c:\users\yyyyyyyyyy\appdata\local\programs\python\python38-32\lib\site-packages\pkg_resources\__init__.py", line 2443, in load
    return self.resolve()
  File "c:\users\yyyyyyyyyy\appdata\local\programs\python\python38-32\lib\site-packages\pkg_resources\__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\yyyyyyyyyy\appdata\local\programs\python\python38-32\lib\site-packages\gw2pvo\__main__.py", line 9, in <module>
    from astral import Astral
ImportError: cannot import name 'Astral' from 'astral' (c:\users\yyyyyyyyyy\appdata\local\programs\python\python38-32\lib\site-packages\astral\__init__.py)

robroy11, 2020-03-01:
Have same problem. Any solution?

svdgraaf, 2020-03-21:
The "problem" is that Astral released a version 2: https://pypi.org/project/astral/#history, and gw2pvo is not compatible with v2 yet. You can force the install of version 1.x with `sudo pip install astral==1.10.1`, which will install the (currently) latest 1.x release.

svdgraaf, 2020-03-21:
You might have to uninstall astral v2 first btw: `sudo pip uninstall astral`

R0yk3, 2020-03-23:
Thankyou This works for me..

DynamicDreams, 2020-04-03:
I'm experiencing the same issue. I did first:
sudo pip uninstall astral

Then:
sudo pip install astral==1.10.1

But I'm still having this error:
ImportError: cannot import name 'Astral' from 'astral' (/usr/local/lib/python3.7/dist-packages/astral/__init__.py)

What else can I try?

walterwhite81, 2020-05-11:
@DynamicDreams manually install the package worked for me:

pip3 install https://pypi.lcsb.uni.lu/packages/86/05/25c772065bb6384789ca0f6ecc9d0bdd0bc210064e5c78453ee15124082e/astral-1.10.1.tar.gz#sha256=d2a67243c4503131c856cafb1b1276de52a86e5b8a1d507b7e08bee51cb67bf1

this installs astral 1.10.1 from source https://pypi.lcsb.uni.lu/simple/astral/

Merlin1201, 2020-05-24:
To bad it's not working for me.  dam using a raspberry pi 3 with p1monitor as image.
The older img are working fine but after building a new one its not working.

I did try installing but still this error
 sudo pip install astral==1.10.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: astral==1.10.1 in /usr/local/lib/python2.7/dist-packages (1.10.1)
Requirement already satisfied: pytz in /usr/local/lib/python2.7/dist-packages (from astral==1.10.1) (2018.3)
pygobject 3.30.4 requires pycairo>=1.11.1, which is not installed.

May 24 13:35:09 p1mon gw2pvo[300]:     from astral import Astral
May 24 13:35:09 p1mon gw2pvo[300]: ImportError: cannot import name 'Astral' from 'astral' (/usr/local/lib/python3.7/dist-packages/astral/__init__.py)
May 24 13:35:09 p1mon systemd[1]: gw2pvo.service: Main process exited, code=exited, status=1/FAILURE
May 24 13:35:09 p1mon systemd[1]: gw2pvo.service: Failed with result 'exit-code'

walterwhite81, 2020-05-24:
Try calling it with Python2 instead of python3.

Sent from ProtonMail Mobile

Merlin1201, 2020-05-24:
i have no idea what you mean, ,any an example?

sam-utr, 2020-06-26:
I am facing the same issue when trying to setup this program on my Rpi 4. I tried to uninstall and reinstall astral 1.1 but still getting the same error:

Traceback (most recent call last):
  File "/usr/local/bin/gw2pvo", line 6, in <module>
    from gw2pvo.__main__ import run
  File "/usr/local/lib/python3.7/dist-packages/gw2pvo/__main__.py", line 9, in <module>
    from astral import Astral
ImportError: cannot import name 'Astral' from 'astral' (/usr/local/lib/python3.7/dist-packages/astral/__init__.py)

Is there any other solution or workaround for this?

eNeRGy164, 2020-06-26:
Sorry I  the code changes in #43

eNeRGy164, 2020-06-28:
gw2pvo-docker 1.3.0 runs into the same issues with Astral.

> ImportError: cannot import name 'Astral'

@markruys can we either:
- Upgrade code to Astral v2.x (#43)
- Have a trick so gw2pvo forces pip to resolve to Astral v1.x

As I'm a noob for Python, I really don't know how to do that last one.

Merlin1201, 2020-06-28:
For now that 1.3.0 version is running again on my older p1mon distro.
But when I upgrade to a newer p1mon distro, iam still having that astral isue.

p1mon@p1mon(eth0=192.168.1.140 wlan0=192.168.1.162):~ $ sudo pip install https://github.com/sffjunkie/astral/archive/1.10.1.tar.gz
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/sffjunkie/astral/archive/1.10.1.tar.gz
  Downloading https://github.com/sffjunkie/astral/archive/1.10.1.tar.gz
     | 112kB 354kB/s
Requirement already satisfied: pytz in /usr/local/lib/python2.7/dist-packages (from astral==1.10.1) (2018.3)
Building wheels for collected packages: astral
  Running setup.py bdist_wheel for astral ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-zcwVHe/wheels/3b/c4/65/0964f15c8e00cc7e086f50d9d4533f10e06ede912d3f545124
Successfully built astral
pygobject 3.30.4 requires pycairo>=1.11.1, which is not installed.
Installing collected packages: astral
Successfully installed astral-1.10.1
You are using pip version 10.0.1, however version 20.2b1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

markruys, 2020-06-28:
I've merged #43, so now we're using Astral 2 and it should work now.

Merlin1201, 2020-06-28:
I installed 1.3.0 again, sudo pip3 install https://github.com/markruys/gw2pvo/releases/download/1.3.0/gw2pvo-1.3.0.tar.gz 
stil getting that same error.

Jun 28 18:24:17 p1mon systemd[1]: Stopped Read GoodWe inverter and upload data to PVOutput.org.
Jun 28 18:24:17 p1mon systemd[1]: Started Read GoodWe inverter and upload data to PVOutput.org.
Jun 28 18:24:18 p1mon gw2pvo[1147]: Traceback (most recent call last):
Jun 28 18:24:18 p1mon gw2pvo[1147]:   File "/usr/local/bin/gw2pvo", line 6, in <module>
Jun 28 18:24:18 p1mon gw2pvo[1147]:     from gw2pvo.__main__ import run
Jun 28 18:24:18 p1mon gw2pvo[1147]:   File "/usr/local/lib/python3.7/dist-packages/gw2pvo/__main__.py", line 9, in <module>
Jun 28 18:24:18 p1mon gw2pvo[1147]:     from astral import Astral
Jun 28 18:24:18 p1mon gw2pvo[1147]: ImportError: cannot import name 'Astral' from 'astral' (/usr/local/lib/python3.7/dist-packages/astral/__init__.py)
Jun 28 18:24:18 p1mon systemd[1]: gw2pvo.service: Main process exited, code=exited, status=1/FAILURE
Jun 28 18:24:18 p1mon systemd[1]: gw2pvo.service: Failed with result 'exit-code'.

markruys, 2020-06-28:
Yes, that was to be expected as 1.3.0 does not contain the commits from #43. You have to wait for the next release, but that shouldn't take long.

markruys, 2020-06-28:
I've just released it as a new package:
```
pip3 install https://github.com/markruys/gw2pvo/releases/download/1.3.1/gw2pvo-1.3.1-py3-none-any.whl
```

closed 2020-06-28 by markruys

Merlin1201, 2020-06-28:
It's working perfect, thank you very much.
I'm using this script on a raspberry with the img p1monitor and I was stuck with the older version of p1mon and it's working now.
again thank you for your script, couse it's the only one that works for me.

jeffheaton, 2022-09-17:
Suggest getting a more standard just pip install astral type install... Moving on to another packages, I don't want to be leading my students through this.