markruys/gw2pvo #40: Dark Sky API should be replaced

franksnl, 2020-06-11:
Dark Sky has joined Apple and does not accept new API signups.

"Our API service for existing customers is not changing today, but we will no longer accept new signups. The API will continue to function through the end of 2021."

https://blog.darksky.net/dark-sky-has-a-new-home/

eNeRGy164, 2020-06-11:
I was already afraid this topic would come up. :)

I really like the Dark Sky API, but that's probably also the reason why Apple bought them.

Does anyone have suggestions for a replacement API?

franksnl, 2020-06-14:
I guess most accurate will be your own or a nearby Netatmo weatherstation https://dev.netatmo.com/apidocumentation/weather

markruys, 2020-06-29:
Netatmo looks promising. You need to create a free account and an app. Then we can use https://pypi.org/project/netatmo/ to query a weather station. I don't know yet how you can get such an id nearby your PV.

franksnl, 2020-06-30:
https://dev.netatmo.com/apidocumentation/weather#getpublicdata
The response based on square lat/lon of your area contains nearby id's.

Looks like function getmeasure required both device_id and module_id.

markruys, 2020-06-30:
Yes, I was thinking in the same direction. Pick a square area, enlarge it until you find weather stations and then pick the closest. With a command line option to overrule this selection mechanism and use a fixed device id. 

I don't think we need to set a module id, we get all sensor data and filter out the temperature.

BTW, there are a few more Python packages implementing the Netamo API, so we should pick the best/most mature.

markruys, 2020-06-30:
I've create a feature branch with a first attempt: https://github.com/markruys/gw2pvo/tree/feature/netatmo. It supports only public Netamo weather stations, so it won't work to obtain historical temperature readings. Please let me know how it works for you.

sam-utr, 2020-07-01:
Didnt try the netatmo branch yet but another option could be openweather. See https://openweathermap.org/darksky-openweather

Will also try out the new feature branch in the meanwhile.

franksnl, 2020-07-02:
I've tested the feature branch and works like expected! Leaving device-id empty uses a public weatherstation close to my location. Looks like my own device is excluded from the getpublicdata response. Using the MAC address at device-id works for using my own device.

markruys, 2020-07-05:
But I've merged the Netatmo into trunk and released it as 1.3.4. You can try it like:

```
sudo pip3 install gw2pvo==1.3.4
```

willlllzy, 2020-10-30:
I just added Netatmo to the config, it looks like the keys are correct and connection established to the API however the log is showing error and it doesn't upload to PVoutput:
 
INFO     Normal, 794.0 W now, 21.9 kWh today, 83.2 kWh all time, 243.7 V grid, 617.8 V PV
ERROR    'city'

L-four, 2020-10-30:
My inverter reports it's own temperature I have been logging that: https://github.com/L-four/gw2pvo/commit/74d46401979ed3f0ffa0f5876ffaeb1878d7be1b

ThinkPadNL, 2021-02-10:
Why is a weather API needed for `gw2pvo` needed in the first place? PVOutput already supports this by itself. 
Just create an API-key at OpenWeatherMap and fill it in at the PVOutput system settings.

eNeRGy164, 2021-02-10:
> Why is a weather API needed for `gw2pvo` needed in the first place? PVOutput already supports this by itself.
> Just create an API-key at OpenWeatherMap and fill it in at the PVOutput system settings.

Because this was built a year before PVOutput started supporting OpenWeatherMap?