markruys/gw2pvo #44: gw2pvo fails due to certificate expiration

lbbrhzn, 2020-06-25:
The gw2pvo script currently fails due to a certificate expiration:

{code>
WARNING  HTTPSConnectionPool(host='globalapi.sems.com.cn', port=443): Max retries exceeded with url: /api/v1/PowerStation/GetMonitorDetailByPowerstationId (Caused by SSLError(SSLError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:841)'),))
<code>

As a temporary workaround, you can disable verification:

in gw_api.py:
{code}
                r = requests.post(self.base_url + url, verify=False, headers=headers, data=payload, timeout=10)
{code}

eNeRGy164, 2020-06-25:
I already have created a PR for that.

lbbrhzn, 2020-06-25:
great, i'll close this issue

closed 2020-06-25 by lbbrhzn