markruys/gw2pvo #45: GW2PVO error - Certicate verify failed - Max retries exceeded

pspsspeed, 2020-06-26:
Hi, gw2pvo was working fine til yesterday when it stopped updating PVOutput with the error below:

WARNING  HTTPSConnectionPool(host='globalapi.sems.com.cn', port=443): Max retries exceeded with url: /api/v1/PowerStation/GetMonitorDetailByPowerstationId (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')])")))

Were there any changes on SEMS that someone is aware of?

eNeRGy164, 2020-06-26:
They let the certificate expire. But they use a different (v2) api to login with their apps. So I created a #42 for that.

pspsspeed, 2020-06-26:
Thank you for the reply - will wait for update

walterwhite81, 2020-06-26:
works like a charm! thanks @eNeRGy164

Ralf7, 2020-06-27:
Good to read that there is a solution. I informed GoodWe about it and they replied was that the API is only for business customers ......
I’m looking forward to an update of the script from Mark Ruys. @markruys

markruys, 2020-06-27:
I've release 1.3.0 which incorporates the fix from #42.

closed 2020-06-27 by markruys

pspsspeed, 2020-06-28:
Thank you so much - working 100%

CoenWhite, 2020-06-28:
Thanks for the update, it's working again!

I'm having some trouble updating the old data though... it gives a "ERROR    Failed to call GoodWe API" error, but if I turn on DEBUG info I see information getting through:

```
DEBUG    Starting new HTTPS connection (1): eu.semsportal.com:443
DEBUG    https://eu.semsportal.com:443 "POST /api/PowerStationMonitor/GetPowerStationPowerAndIncomeByDay HTTP/1.1" 200 444
DEBUG    {'hasError': False, 'code': 0, 'msg': 'Success', 'components': {'msgSocketAdr': '', 'api': 'http://eu.semsportal.com:82/api/PowerStationMonitor/GetPowerStationPowerAndIncomeByDay', 'para': None, 'timeSpan': 0, 'langVer': 73}, 'data': [{'i': 6.384, 'p': 30.4, 'd': '06/26/2020'}, {'i': 4.767, 'p': 22.7, 'd': '06/27/2020'}]}
DEBUG    Starting new HTTPS connection (1): semsportal.com:443
DEBUG    https://semsportal.com:443 "POST /api/v2/Common/CrossLogin HTTP/1.1" 200 527
ERROR    Failed to call GoodWe API
WARNING  2020-06-27 - Received bad data {}
```

Is this a related problem?

eNeRGy164, 2020-06-28:
Seems related to #23 and #34