markruys/gw2pvo #46: [not an issue with gw2pvo] globalapi.sems.com.cn certificate expired on Jun 25

Botje, 2020-06-29:
Thank you for writing this handy tool.

I am creating this issue so your users are aware that gw2pvo currently cannot function correctly due to a configuration error at SEMS' side:
```
Certificate:                                                                    
    Data:                                                                       
        Version: 3 (0x2)                                                        
        Serial Number:                                                          
            03:38:c3:39:1f:00:e0:8f:9f:2b:ff:91:c0:bd:14:95                     
    Signature Algorithm: sha256WithRSAEncryption                                
        Issuer: C = US, O = DigiCert Inc, OU = www.digicert.com, CN = Encryptio$
 Everywhere DV TLS CA - G1                                       
        Validity                                                 
            Not Before: Jun 26 00:00:00 2019 GMT                                
            Not After : Jun 25 12:00:00 2020 GMT                               
        Subject: CN = globalapi.sems.com.cn           
```

Until SEMS fixes this on their end, users can work around it by adding `verify=False` to the two `requests.post()` lines in `gw_api.py`.

markruys, 2020-06-29:
Most likely you're still using an old version of gw2pvo. Can you upgrade to 1.3.2 and check again?

Botje, 2020-06-29:
Upgrading to 1.3.2 works, thanks. I had incorrectly assumed the `astral` changes did not change the API server used.

closed 2020-06-29 by Botje