markruys/gw2pvo #47: error after upgrading to 1.3.2.

wiljums, 2020-06-29:
Jun 29 09:32:36 raspberrypi systemd[1]: Started Read GoodWe inverter and upload data to PVOutput.org.
Jun 29 09:32:37 raspberrypi gw2pvo[25284]: Traceback (most recent call last):
Jun 29 09:32:37 raspberrypi gw2pvo[25284]:   File "/usr/local/bin/gw2pvo", line 6, in <module>
Jun 29 09:32:37 raspberrypi gw2pvo[25284]:     from gw2pvo.__main__ import run
Jun 29 09:32:37 raspberrypi gw2pvo[25284]:   File "/usr/local/lib/python3.7/dist-packages/gw2pvo/__main__.py", line 9, in <module>
Jun 29 09:32:37 raspberrypi gw2pvo[25284]:     from astral import LocationInfo
Jun 29 09:32:37 raspberrypi gw2pvo[25284]: **ImportError: cannot import name 'LocationInfo' from 'astral' (/usr/local/lib/python3.7/dist-packages/astral.py)**
Jun 29 09:32:37 raspberrypi systemd[1]: gw2pvo.service: Main process exited, code=exited, status=1/FAILURE
Jun 29 09:32:37 raspberrypi systemd[1]: gw2pvo.service: Failed with result 'exit-code'.

What is causing this error?

wiljums, 2020-06-29:
Upgraded Astral to 2.2, now it works.

closed 2020-06-29 by wiljums

markruys, 2020-06-29:
Just curious, how did you install gw2pvo? I would expect it would update Alstral for you automatically.

wiljums, 2020-06-29:
I used to command you advised:

sudo pip3 install
https://github.com/markruys/gw2pvo/releases/download/1.3.2/gw2pvo-1.3.2-py3-none-any.whl

markruys, 2020-06-29:
I would like to reproduce this issue to understand what goes wrong. I get:
```
markr@peat:~$ pip3 install https://github.com/markruys/gw2pvo/releases/download/1.3.2/gw2pvo-1.3.2-py3-none-any.whl
Collecting gw2pvo==1.3.2 from https://github.com/markruys/gw2pvo/releases/download/1.3.2/gw2pvo-1.3.2-py3-none-any.whl
  Downloading https://github.com/markruys/gw2pvo/releases/download/1.3.2/gw2pvo-1.3.2-py3-none-any.whl
Collecting astral (from gw2pvo==1.3.2)
  Downloading https://files.pythonhosted.org/packages/91/60/7cc241b9c3710ebadddcb323e77dd422c693183aec92449a1cf1fb59e1ba/astral-2.2-py2.py3-none-any.whl
Collecting requests (from gw2pvo==1.3.2)
  Downloading https://files.pythonhosted.org/packages/45/1e/0c169c6a5381e241ba7404532c16a21d86ab872c9bed8bdcd4c423954103/requests-2.24.0-py2.py3-none-any.whl (61kB)
    100% |████████████████████████████████| 71kB 5.1MB/s 
Collecting pytz (from astral->gw2pvo==1.3.2)
  Downloading https://files.pythonhosted.org/packages/4f/a4/879454d49688e2fad93e59d7d4efda580b783c745fd2ec2a3adf87b0808d/pytz-2020.1-py2.py3-none-any.whl (510kB)
    100% |████████████████████████████████| 512kB 3.0MB/s 
Collecting chardet<4,>=3.0.2 (from requests->gw2pvo==1.3.2)
  Downloading https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl (133kB)
    100% |████████████████████████████████| 143kB 6.4MB/s 
Collecting certifi>=2017.4.17 (from requests->gw2pvo==1.3.2)
  Downloading https://files.pythonhosted.org/packages/5e/c4/6c4fe722df5343c33226f0b4e0bb042e4dc13483228b4718baf286f86d87/certifi-2020.6.20-py2.py3-none-any.whl (156kB)
    100% |████████████████████████████████| 163kB 7.2MB/s 
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests->gw2pvo==1.3.2)
  Downloading https://files.pythonhosted.org/packages/e1/e5/df302e8017440f111c11cc41a6b432838672f5a70aa29227bf58149dc72f/urllib3-1.25.9-py2.py3-none-any.whl (126kB)
    100% |████████████████████████████████| 133kB 11.8MB/s 
Collecting idna<3,>=2.5 (from requests->gw2pvo==1.3.2)
  Downloading https://files.pythonhosted.org/packages/a2/38/928ddce2273eaa564f6f50de919327bf3a00f091b5baba8dfa9460f3a8a8/idna-2.10-py2.py3-none-any.whl (58kB)
    100% |████████████████████████████████| 61kB 5.6MB/s 
Installing collected packages: pytz, astral, chardet, certifi, urllib3, idna, requests, gw2pvo
Successfully installed astral certifi chardet-2.3.0 gw2pvo idna pytz requests-2.9.1 urllib3-1.13.1
You are using pip version 8.1.1, however version 20.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

As you can see it installed Astral 2.2.

Do you have your install log still at hand? Else, which OS you installed gw2pvo on? Latest Rasbian?

wiljums, 2020-06-29:
I used Putty, but closed it afterwards. How can I retrieve the install log?

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Willem

Ralf7, 2020-06-29:
I am encountering the same problem as @wiljums 
It's not installing the latest astral, so goodwe2pvoutput.service is not starting (it results in an exit code status=1/failure)
I'm not running the latest raspbian. When I use the following comment `cat /etc/debian_version` it results in 9.11
Do I need to upgrade to the lastest Raspbian to run it?
If not, how can I upgrade astral to the desired version 2.2 ?