markruys/gw2pvo #48: Calling with --date fails on upload to PVOutput

tsbkelly, 2020-06-30:
Log as follows:

`DEBUG    Starting new HTTPS connection (1): eu.semsportal.com:443
DEBUG    https://eu.semsportal.com:443 "POST /api/v2/PowerStationMonitor/GetPowerStationPowerAndIncomeByDay HTTP/1.1" 200 445
DEBUG    {'hasError': False, 'code': 0, 'msg': 'Success', 'data': [{'d': '06/10/2020', 'p': 3.8, 'i': 0.646}, {'d': '06/11/2020', 'p': 5.9, 'i': 1.003}], 'components': {'para': None, 'langVer': 73, 'timeSpan': 0, 'api': 'http://eu.semsportal.com:82/api/v2/PowerStationMonitor/GetPowerStationPowerAndIncomeByDay', 'msgSocketAdr': ''}}
DEBUG    {'data': '20200611,04:10,1,6.11;20200611,04:15,1,6.56;20200611,04:20,1,7.78;20200611,04:25,2,11.47;20200611,04:30,3,13.57;20200611,04:35,4,14.53;20200611,04:40,4,15.37;20200611,04:45,6,15.56;20200611,04:50,6,15.92;20200611,04:55,7,16.05;20200611,05:00,8,16.2;20200611,05:05,9,16.57;20200611,05:10,10,16.76;20200611,05:15,11,17.02;20200611,05:20,12,19.64;20200611,05:25,13,22.62;20200611,05:30,15,21.72;20200611,05:35,16,27.98;20200611,05:40,19,39.0;20200611,05:45,21,42.96;20200611,05:50,24,39.18;20200611,05:55,26,36.6;20200611,06:00,28,38.79;20200611,06:05,32,56.84;20200611,06:10,36,81.1;20200611,06:15,41,78.15;20200611,06:20,45,77.4;20200611,06:25,49,66.36;20200611,06:30,53,58.8;20200611,06:35,56,66.4'}
DEBUG    Starting new HTTPS connection (1): pvoutput.org:443
DEBUG    https://pvoutput.org:443 "POST /service/r2/addbatchstatus.jsp HTTP/1.1" 400 None
WARNING  400 Client Error: Bad Request for url: https://pvoutput.org/service/r2/addbatchstatus.jsp
DEBUG    Starting new HTTPS connection (1): pvoutput.org:443
DEBUG    https://pvoutput.org:443 "POST /service/r2/addbatchstatus.jsp HTTP/1.1" 400 None
WARNING  400 Client Error: Bad Request for url: https://pvoutput.org/service/r2/addbatchstatus.jsp
DEBUG    Starting new HTTPS connection (1): pvoutput.org:443
`

markruys, 2020-06-30:
According to https://pvoutput.org/help.html, the date may not be more than 14 days in the past (90 if you're a donator). I'll try to return a more sensible error message instead of Bad Request.

tsbkelly, 2020-06-30:
Ah, that's my fault for not double-checking my commands. Closing now.

closed 2020-06-30 by tsbkelly