markruys/gw2pvo #49: --config not working

sam-utr, 2020-07-01:
I tried to create and use the config file but received the following error:

$ gw2pvo --config /etc/gw2pvo.cfg --log debug
usage: gw2pvo [-h] --gw-station-id ID --gw-account ACCOUNT --gw-password
              PASSWORD [--pvo-system-id ID] [--pvo-api-key KEY]
              [--pvo-interval {5,10,15}] [--darksky-api-key DARKSKY_API_KEY]
              [--log {debug,info,warning,critical}] [--date YYYY-MM-DD]
              [--pv-voltage] [--skip-offline] [--city CITY] [--csv CSV]
              [--version]
gw2pvo: error: the following arguments are required: --gw-station-id, --gw-account, --gw-password

markruys, 2020-07-01:
Most likely you installed the latest release. I haven't released this change yet, so if you want to use it, you must clone the git repo yourself and use that gw2pvo source.

markruys, 2020-07-01:
Another cause could be that the config file is missing these settings. Are you using underscores instead of dashes as in the example in README.md? Can you share us your config file?

sam-utr, 2020-07-01:
> Most likely you installed the latest release. I haven't released this change yet, so if you want to use it, you must clone the git repo yourself and use that gw2pvo source.

Yes, indeed I installed the latest release. That is the cause of the problem then. I will close the issue.

closed 2020-07-01 by sam-utr

eNeRGy164, 2020-07-01:
+1 for enthusiasm