markruys/gw2pvo #50: problem with config file

JWDebler, 2020-07-05:
Hi,
I tried to use the config file as described, but when running the command I get following error:

```
odroid@odroid:~$ gw2pvo --config gw2pvo.cfg --date 2020-06-25
usage: gw2pvo [-h] --gw-station-id ID --gw-account ACCOUNT --gw-password
PASSWORD [--pvo-system-id ID] [--pvo-api-key KEY]
[--pvo-interval {5,10,15}] [--darksky-api-key DARKSKY_API_KEY]
[--log {debug,info,warning,critical}] [--date YYYY-MM-DD
--pv-voltage] [--skip-offline] [--city CITY] [--csv CSV]                                                                                                                                                                                    [--version]
gw2pvo: error: the following arguments are required: --gw-station-id, --gw-account, --gw-password
```
I am on version 1.3.3, installed via pip as described.

My config file looks like this, where 'xxx' stands for the respective values. I tried using quotes around them but that lead to the same result. Running gw2pvo with all the flags on the command line works. 

```
[Defaults]
gw_station_id = xxx
gw_account = xxx
gw_password = xxx
pvo_api_key = xxx
pvo_system_id = xxx
darksky_api_key = xxx
```

Cheers

markruys, 2020-07-05:
This feature is only available as a feature branch. It's not part of release 1.3.3.

But I've just merged this feature and released it as 1.3.4. You can try:

```
sudo pip3 install gw2pvo==1.3.4
```

JWDebler, 2020-07-05:
Thanks. Now I get this error though:

```
Collecting gw2pvo==1.3.4
Downloading https://files.pythonhosted.org/packages/12/93/4cd45fdd9ef30f72c96433224d40b4d6d369eaacff21c96c2f9a96abf258/gw2pvo-1.3.4-py3-none-any.whl
Requirement already satisfied: astral>=2 in /usr/local/lib/python3.6/dist-packages (from gw2pvo==1.3.4)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from gw2pvo==1.3.4)
Requirement already satisfied: pytz in /usr/lib/python3/dist-packages (from astral>=2->gw2pvo==1.3.4)
Requirement already satisfied: dataclasses; python_version == "3.6" in /usr/local/lib/python3.6/dist-packages (from astral>=2->gw2pvo==1.3.4)
Installing collected packages: gw2pvo
Found existing installation: gw2pvo 1.3.3
Uninstalling gw2pvo-1.3.3:
Successfully uninstalled gw2pvo-1.3.
Successfully installed gw2pvo-1.3.4

odroid@odroid:~$ gw2pvo --version
Traceback (most recent call last):
File "/usr/local/bin/gw2pvo", line 7, in <module>
from gw2pvo.__main__ import run
File "/usr/local/lib/python3.6/dist-packages/gw2pvo/__main__.py", line 20, in <module>
from gw2pvo import netatmo_api 
File "/usr/local/lib/python3.6/dist-packages/gw2pvo/netatmo_api.py", line 4, in <module>
from oauthlib.oauth2 import LegacyApplicationClient
ModuleNotFoundError: No module named 'oauthlib'
```

markruys, 2020-07-05:
Hm, i missed a new dependency. New release is needed.

markruys, 2020-07-05:
Should add some kind of unit/QC testing, now I need too many releases to get it stable. Anyway, @JWDebler, 1.3.5 is out and should fix your problem.

JWDebler, 2020-07-06:
Thanks @markruys , everything works now!

closed 2020-07-06 by JWDebler