markruys/gw2pvo #52: Consumption not showing with Goodwe HomeKiy

isbrodie1, 2020-10-25:
Hi all, 

Just confirming that it doesn't seem to scrape consumption, it's on the SEMS portal but not moving to PVoutput
Cheers

jakezp, 2020-10-28:
Nope... No support for consumption... I have also been looking for a way to upload consumption data. I did references to it being added in an [old pull request](https://github.com/markruys/gw2pvo/pull/24), but it never got merged. 

I did mess around adding consumption (with very limited python knowledge) and it is working fine for my personal setup.... If you want to [check it out](https://github.com/jakezp/gw2pvo-alt/)... I'm not making any claims that it will work for you, nor am I able to offer any support or assistance if it doesn't work. Have a look and if you're feeling adventurous, give it go. I've also added support for openweather as a replacement for darksky.

darek-margas, 2021-06-04:
I made it working but didn't do much for backward compatibility (say, no checks if there is homekit installed and so on). Works for updates loop, does not for historical data as I haven't found any API call to extract load history.

darek-margas, 2021-11-17:
Is anyone still interested?

darek-margas, 2021-11-18:
> 
> 
> Yep, but I don't think the inverter shows consumption - how can it?

Needs Homekit installed, as per subject.

wifievangelist, 2022-07-16:
@darek-margas : Hi Darek. Have you managed to get consumption with HomeKit working? I have just got a GoodWe/HomeKit combo installed. I can push the data to PVOutput but cannot see consumption. I am using everything at default and using following command:
gw2pvo --gw-station-id XXXXX --gw-account XXXXX --gw-password XXXXX --pvo-system-id XXX --pvo-api-key XXXXX --log debug

darek-margas, 2022-07-23:
> @darek-margas : Hi Darek. Have you managed to get consumption with HomeKit working? I have just got a GoodWe/HomeKit combo installed. I can push the data to PVOutput but cannot see consumption. I am [...]

Yes, it worked for me long time. I replaced monitoring by iotawatt but still have test site using this. 
[https://pvoutput.org/list.jsp?id=97109&sid=85607](url)

lyricnz, 2022-07-28:
@darek-margas can you tell me about IoTaWatt vs goodwe homekit?  Should I preference one over the other?  (I am computer+coding technical, but new to solar).  Was going to contribute to your fork, if that's the most active/maintained one around...

darek-margas, 2022-07-29:
> @darek-margas can you tell me about IoTaWatt vs goodwe homekit? Should I preference one over the other? (I am computer+coding technical, but new to solar). Was going to contribute to your fork, if that's the most active/maintained one around... Friend also has unused https://www.flukso.net/ he's happy to give me

I am not a fond of cloud stuff as it does not work due to issues beyond my control. This is why I got Iotawatt. 
What I got better this way is:
- It is local, has some UI and good REST API
- I use it with Home Assistant so that everything works even with no internet
- It has built in uploader for PVoutput
- I calibrated it so that discrepancy between PVoutput and official meter is below 5c per day
- It has number of ports so that I was able to put CT on every circuit giving me a lot better picture than just grid and solar view
- It is pretty much open source so you can make adjustments and compile or easily make it your own Tasmota/ESPhome thing.

What was/is missing in it:
- For PVoutput upload, if you need SSL you need a proxy as ESP has not enough CPU power
- It did not provide aggregation. It has been added recently but it has impact on SD card and also reducing free memory causing reboots. Lack of aggregation is not an issue with Home Assistant but when I looked at making my own uploader to ChargeHQ, it seems easier from Goodwe cloud as there is aggregation.

lyricnz, 2022-07-29:
100% agree on local vs cloud; plus I had to setup guest wifi, as I don't trust Goodwe.  So you're still using semsportal for the inverter generation data? 

Happy to leave embedded-devices doing as small a job as possible, and aggregate elsewhere (on my pi etc)

Noob question - which wires do the sensors clip to (relative to "feed in", "inverter", and "feed to house")?   I have three phase feed-in also, but not using any three-phase devices yet.

darek-margas, 2022-07-29:
> 100% agree on local vs cloud; plus I had to setup guest wifi, as I don't trust Goodwe. So you're still using semsportal for the inverter generation data?

I use separate IoT wifi networks for all IoT devices including Goodwe. I query inverter directly using modbus over IP. This was developed as python module and then integrated into HA. However, this is only to see how inverter works (temps, string voltage, etc) as it does not provide more than just its output which is also incorrect at low levels. For all other monitoring I read Iotawatt. It can do 3P, just taakes 3 ports for 3x CT and then, if you need correct readings, 3x reference voltage. 6 ports gone but still a few to clamp elsewhere. You would have to clip all 3 grid live wires, use 3 reference transformers (or one and emulate 2), inverter live wires (1 or 3 depending on inverter) and then whatever else you want to measure within house.