markruys/gw2pvo #55: Add Temperature to the upload data

muzzak123, 2021-01-08:
Hi, just got my new system installed and this is perfect to upload my Goodwe data to PVOutput. Thanks for doing this. It was so easy to install and great detailed instructions. 
Just a comment/improvement. My inverter produces temperature data and I can't see anyway to upload it using gw2pvo. Is there a config I can change to do this ?

eNeRGy164, 2021-01-08:
The current design is that the temperature coming from the invertor is not the outside ambient temperature, as intended by PVOutput, but the temperature of the inverter itself, which is much higher.
Therefore, there is only the option to configure a Weather API to get the ambient temperature, and no option to pass the inverter temperature.

muzzak123, 2021-01-09:
I'm pretty sure you can use either. See this post [Temperature Displaying] https://forum.pvoutput.org/t/temperature-displaying/532
The last bit says ...

> To use weather temperature then disable the solaredge temperature as per the above setting.
To use inverter temperature then remove the weather station details.

muzzak123, 2021-01-09:
Doesn't matter. I've found another way to do it. Thanks

closed 2021-01-09 by muzzak123