markruys/gw2pvo #59: Is GoodWe API still working?

vijghph, 2021-11-21:
I was regularly getting timeouts or an indication that GetPowerStationPowerAndIncomeByDay() Received bad data
...but usually this was still working by repeatedly trying. 

Today I cannot manage to get data at all. I am afraid the GoodWe API that my (tuned) version of gw2pvo is using has been deprecated by GoodWe, isn't it?

vijghph, 2021-11-21:
I was still using a customer version 1.2.1

I upgraded to 1.3.6 and it seems to work and not requiring my former fix anymore (multiple inverters setup now ok)

Steve-Wax, 2021-11-21:
I had the same problem with GetPowerStationPowerAndIncomeByDay().  I'm using the following:

I'm using this for the token:
r = requests.post(self.global_url + 'v2/Common/CrossLogin', headers=headers, data=loginPayload, timeout=10)
And this for the call:
headers = { 'User-Agent': 'SEMS Portal/3.1 (iPhone; iOS 13.5.1; Scale/2.00)', 'Token': self.token }

What are you putting in the header to make it work?

vijghph, 2021-11-22:
I did use pip to upgrade and it is indeed using code implant to what you report.

see https://github.com/markruys/gw2pvo/blob/master/gw2pvo/gw_api.py