markruys/gw2pvo #66: Is fork still maintained?

lyricnz, 2022-07-28:
Looking at this fork, based on a lead from forum.pvoutput.org, but it has some bugs (especially around case where consumption data is not available).  Before I fix these, and raise PR, is this the right/best fork to do it?  (no point if not maintained)

eg:
data items are 'None' not None
exceptions gobbled without way to debug
incorrect variable name referenced in error case

elRadix, 2022-08-03:
indeed, which is the one to use ?

lyricnz, 2022-09-19:
I've switched to using Home Assistant, so no longer use gw2pvo.

markruys, 2022-10-18:
I've switched to Home Assistant too and don't use this project anymore. So I guess we need another maintainer, or archive this project.

elRadix, 2022-10-18:
> I've switched to Home Assistant too and don't use this project anymore. So I guess we need another maintainer, or archive this project.

How do you send data to pvoutput.org through home assistant
can you share your config example please

markruys, 2022-10-18:
> How do you send data to pvoutput.org through home assistant can you share your config example please

I've stopped using PVOutput. Home Assistant does have a PVOutput integration. Haven't tried it myself.

lyricnz, 2022-10-18:
This is completely off-topic, but...

I don't use any "integration" as such. In HA configuration.yaml I created a custom REST command that POSTs the data to pvoutput API. See https://pvoutput.org/help/api_specification.html#add-status-service for docs.

The command copies a few HA sensors (which you set up before) into a few of the pvoutput values.  It also uses a couple of "secrets" so I don't have to put my pvo user/password in there.

```
rest_command:
  pvoutput_generation:
    method: POST
    url: https://pvoutput.org/service/r2/addstatus.jsp
    headers:
      X-Pvoutput-Apikey: !secret pvo_api_key
      X-Pvoutput-SystemId: !secret pvo_system_id
    payload: 'd={{now().strftime("%Y%m%d")}}&t={{now().strftime("%H:%M")}}&v1={{states("sensor.today_s_pv_generation") |round(1)*1000}}&v2={{states("sensor.pv_power") |round(0)}}&v3={{states("sensor.main_wh") |round(0)}}&v4={{states("sensor.main") |round(0)}}&v5={{states("sensor.openweathermap_temperature")}}&v6={{states("sensor.on_grid_l1_voltage")}}&v7={{states("sensor.inverter_temperature")}}'
    content_type: "application/x-www-form-urlencoded"
```

My secrets.yaml contains:
```
# Use this file to store secrets like usernames and passwords.
# Learn more at https://www.home-assistant.io/docs/configuration/secrets/
some_password: welcome
pvo_api_key: long_hex_thing
pvo_system_id: pv_system_id
```

HTH

lyricnz, 2022-12-02:
Closing this issue with a resolution of "nope"

closed 2022-12-02 by lyricnz

vijghph, 2023-04-07:
> > How do you send data to pvoutput.org through home assistant can you share your config example please
> 
> I've stopped using PVOutput. Home Assistant does have a PVOutput integration. Haven't tried it myself.

Unfortunately PVOutput integration requires a local access to the Goodwe inverters ...which themselves need to be upgraded to the last firmware or otherwise are not supported. Not a trivial thing.

I preferred a lot the gw2pvo approach which worked seamlessly until very recently.

@markruys it would be great if you could validate the OpenWeather integration that some have contributed to among the PullRequests.