markruys/gw2pvo #68: syntax error

varkanoid, 2022-12-08:
Hi,

sorry see the attached picture. I`m a right noob what am I doing wrong ?

regards

Jools

[image]
[image]

eNeRGy164, 2022-12-08:
You don't have pip installed

varkanoid, 2022-12-08:
I enabled it at install ? 
[image]

varkanoid, 2022-12-08:
I`ll figure it out I wont take up anymore of your time.

closed 2022-12-08 by varkanoid