markruys/gw2pvo #69: 403 Client Error: Forbidden for url

pumpkins001, 2022-12-12:
Anyone else getting this error? Appears to have stopped working 28th Nov 2022.

403 Client Error: Forbidden for url: https://semsportal.com/api/v1/PowerStation/GetMonitorDetailByPowerstationId
ERROR    Failed to call GoodWe API
ERROR    'inverter'

Made no changes to script. Data is present in SEMS portal.

pumpkins001, 2022-12-20:
I've found the issue, SEMS have updated the API so the url is now v2 instead of v1. I've updated this in the respective file locations, but is anyone able to tell me how to "recompile" as such so that when it runs it uses v2 instead of v1?

Janvier123, 2023-01-17:
I can make a fork and update it if you want

pumpkins001, 2023-01-17:
That would be awesome.

wiljums, 2023-01-17:
API V2 has already been integrated?
[https://github.com/markruys/gw2pvo/pull/42](url)

Janvier123, 2023-01-28:
> That would be awesome.
> […](#)

Sorry i did not recive any messages from github, i think ive fixed that,
Anyway ive found a better script that does the job ALLOT faster and more reliable using UDP protocol instead of the sems API

https://github.com/ASlatius/gw2mqtt-pvo

Ill fork that one, remove the MQTT and update all the info