markruys/gw2pvo #73: did sems portal data change?

JWDebler, 2023-05-02:
Hi, I know this isn't maintained anymore, but maybe someone else is experiencing the same problem.

As of April 1st 2023 I am not getting any uploads to pvoutput.org anymore.

Checking the log is see lots of these errors:

```
May 02 11:50:14 odroid gw2pvo[22388]: INFO     Normal, 2810.0 W now, 6.9 kWh today, 34268.1 kWh all time, 246.5 V grid, 561.6 V PV
May 02 11:50:19 odroid gw2pvo[22388]: ERROR    Expecting value: line 2 column 1 (char 1)
May 02 11:55:17 odroid gw2pvo[22388]: INFO     Normal, 2815.0 W now, 7.2 kWh today, 34268.4 kWh all time, 246.8 V grid, 561.2 V PV
May 02 11:55:23 odroid gw2pvo[22388]: ERROR    Expecting value: line 2 column 1 (char 1)
```
It seems to be reading data correctly, but then something fails.
Did GoodWe change anything in the data that is expected by this script?

Maybe anyone with more python knowledge than myself knows what to do here.

Cheers

joffa101, 2023-05-02:
It is still working for me.

joffa101, 2023-05-02:
May 02 13:26:26 lplex gw2pvo[1020]: INFO     Normal, 807.0 W now, 4.1 kWh today, 10592.4 kWh all time, 238.4 V grid, 558.6 V PV
May 02 13:31:26 lplex gw2pvo[1020]: INFO     Normal, 994.0 W now, 4.2 kWh today, 10592.5 kWh all time, 238.2 V grid, 572.9 V PV
May 02 13:36:26 lplex gw2pvo[1020]: INFO     Normal, 1048.0 W now, 4.3 kWh today, 10592.6 kWh all time, 238.9 V grid, 572.3 V PV
May 02 13:41:26 lplex gw2pvo[1020]: INFO     Normal, 1603.0 W now, 4.3 kWh today, 10592.6 kWh all time, 240.6 V grid, 574.1 V PV
May 02 13:46:27 lplex gw2pvo[1020]: INFO     Normal, 1017.0 W now, 4.5 kWh today, 10592.8 kWh all time, 237.9 V grid, 557.6 V PV
May 02 13:51:26 lplex gw2pvo[1020]: INFO     Normal, 760.0 W now, 4.5 kWh today, 10592.8 kWh all time, 233.4 V grid, 533.6 V PV
May 02 13:56:26 lplex gw2pvo[1020]: INFO     Normal, 843.0 W now, 4.6 kWh today, 10592.9 kWh all time, 236.5 V grid, 561.8 V PV
May 02 14:01:26 lplex gw2pvo[1020]: INFO     Normal, 419.0 W now, 4.7 kWh today, 10593.0 kWh all time, 235.3 V grid, 537.9 V PV
May 02 14:06:26 lplex gw2pvo[1020]: INFO     Normal, 1332.0 W now, 4.7 kWh today, 10593.0 kWh all time, 237.2 V grid, 581.5 V PV
May 02 14:11:26 lplex gw2pvo[1020]: INFO     Normal, 361.0 W now, 4.8 kWh today, 10593.1 kWh all time, 235.3 V grid, 524.4 V PV

mike-trewartha, 2023-05-02:
It's working for me also

JWDebler, 2023-05-02:
hmmm, that's odd...
restarted but still seeing the same behaviour.

JWDebler, 2023-05-02:
Got it, forgot to also update the new api key in the service file and removing the darksky api.

closed 2023-05-02 by JWDebler